jasoncoon/esp32-fastled-webserver #1: Not serving static files

jasoncoon, 2018-02-16:
I am not currently able to get the ESP32 web server to serve static files like I was able to with the ESP8266.  The browser requests a file fine, and the response from the ESP32 web server has the correct headers, but the response body is empty.

It is able to connect to wifi, and basic HTTP requests work fine, just not serving static files from SPIFFS.

I have tried three different ESP32 web server libraries, with varying degrees of success:

* https://github.com/esp8266/ESPWebServer
* https://github.com/zhouhan0126/WebServer-esp32
* https://github.com/bbx10/WebServer_tng

I'm using this Arduino plugin to upload the sketch data directory containing the static files required for the web app:

[Arduino ESP32 filesystem uploader](https://github.com/me-no-dev/arduino-esp32fs-plugin)

I am able to list the files stored in SPIFFS to the serial monitor:

```
Listing directory: /
  FILE: /css/styles.css  SIZE: 32
  FILE: /favicon.ico  SIZE: 4286
  FILE: /images/atom196.png  SIZE: 5469
  FILE: /index.htm  SIZE: 10886
  FILE: /js/app.js  SIZE: 13430
```

Here's a request for `/index.htm`:

```
GET /index.htm HTTP/1.1
Host: 192.168.86.44
Connection: keep-alive
Pragma: no-cache
Cache-Control: no-cache
Upgrade-Insecure-Requests: 1
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/63.0.3239.132 Safari/537.36
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,image/apng,*/*;q=0.8
DNT: 1
Accept-Encoding: gzip, deflate
Accept-Language: en-US,en;q=0.9
```

And the response headers:
```
HTTP/1.1 200 OK
Content-Type: text/html
Cache-Control: max-age=86400
Content-Length: 10886
Connection: close
```

The response body is empty.

HTTP REST requests work fine.

Here's the REST request:

```
GET /all HTTP/1.1
Host: localhost:8080
Connection: keep-alive
Pragma: no-cache
Cache-Control: no-cache
Accept: */*
X-Requested-With: XMLHttpRequest
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/63.0.3239.132 Safari/537.36
DNT: 1
Referer: http://localhost:8080/index.htm
Accept-Encoding: gzip, deflate, br
Accept-Language: en-US,en;q=0.9
```

and the response:

```
HTTP/1.1 200 OK
server: ecstatic-1.4.1
content-type: text/json
content-length: 520
connection: close
Date: Fri, 16 Feb 2018 02:53:27 GMT

[{"name":"power","label":"Power","type":"Boolean","value":1},{"name":"brightness","label":"Brightness","type":"Number","value":32,"min":1,"max":255},{"name":"pattern","label":"Pattern","type":"Select","value":2,"options":["rainbow","rainbowWithGlitter","confetti","sinelon","juggle","bpm"]},{"name":"autoplaySection","label":"Autoplay","type":"Section"},{"name":"autoplay","label":"Autoplay","type":"Boolean","value":1},{"name":"autoplayDuration","label":"Autoplay Duration","type":"Number","value":2,"min":0,"max":255}]
```

[index.har.zip](https://github.com/jasoncoon/esp32-fastled-webserver/files/1731596/index.har.zip)

jasoncoon, 2018-02-17:
I used Yves BAZIN's fix here: https://github.com/hpwit/lib
from this conversation: https://plus.google.com/+JasonCoon1/posts/UV9YQzXN6a4

Copy `libspiffs.a` to the ESP32 tools directory:
`\Documents\Arduino\hardware\espressif\esp32\tools\sdk\lib\libspiffs.a`

And copy `esp32fs.jar` to the Arduino tools directory:
`\Documents\Arduino\tools\ESP32FS\tool\esp32fs.jar`

closed 2018-02-17 by jasoncoon

marcmerlin, 2019-11-04:
Do you know if this fix ever made it upstream? Copying around a precompiled file from somewhere with no history, no indication of whether tip of tree is better/more recent and so forth, is not super inspiring :)
@hpwit do you know by any chance?

jasoncoon, 2019-11-04:
Good question.  I haven't tried the standard one included in the esp32 sdk in a while.  I would hope it's been fixed.

rorosaurus, 2019-11-09:
Hmm, I wasn't able to get it to work - with or without the fix applied. My browser just seems to sit loading a white page. Maybe I have some strange network settings.
In Chrome console, I see `GET http://192.168.1.203/ net::ERR_CONTENT_LENGTH_MISMATCH 200 (OK)`

FryAllenEthan, 2020-01-01:
I've also tried both versions, fix and no fix, and get the same length mismatch error