jasoncoon/esp32-fastled-webserver #4: Compiling issue

jonny190, 2018-04-01:
`Arduino: 1.8.5 (Linux), Board: "ESP32 Dev Module, QIO, 80MHz, 4MB (32Mb), 921600, None"

/usr/share/arduino/arduino-builder -dump-prefs -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/jonny/.arduino15/packages -hardware /home/jonny/Arduino/hardware -tools /usr/share/arduino/tools-builder -tools /home/jonny/.arduino15/packages -libraries /home/jonny/Arduino/libraries -fqbn=espressif:esp32:esp32:FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10805 -build-path /tmp/arduino_build_786844 -warnings=none -build-cache /tmp/arduino_cache_212744 -prefs=build.warn_data_percentage=75 -verbose /home/jonny/esp32-fastled-webserver/esp32-fastled-webserver.ino
/usr/share/arduino/arduino-builder -compile -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/jonny/.arduino15/packages -hardware /home/jonny/Arduino/hardware -tools /usr/share/arduino/tools-builder -tools /home/jonny/.arduino15/packages -libraries /home/jonny/Arduino/libraries -fqbn=espressif:esp32:esp32:FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10805 -build-path /tmp/arduino_build_786844 -warnings=none -build-cache /tmp/arduino_cache_212744 -prefs=build.warn_data_percentage=75 -verbose /home/jonny/esp32-fastled-webserver/esp32-fastled-webserver.ino
Using board 'esp32' from platform in folder: /home/jonny/Arduino/hardware/espressif/esp32
Using core 'esp32' from platform in folder: /home/jonny/Arduino/hardware/espressif/esp32
Detecting libraries used...
"/home/jonny/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/config" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/bluedroid" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/app_trace" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/app_update" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/bootloader_support" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/bt" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/driver" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/esp32" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/esp_adc_cal" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/ethernet" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/fatfs" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/freertos" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/heap" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/jsmn" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/log" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/mdns" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/mbedtls" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/mbedtls_port" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/nvs_flash" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/openssl" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/spi_flash" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/sdmmc" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/spiffs" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/tcpip_adapter" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/ulp" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/vfs" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/wear_levelling" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/xtensa-debug-module" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/coap" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/console" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/expat" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/json" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/lwip" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/nghttp" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/soc" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/wpa_supplicant" -std=gnu++11 -fno-exceptions -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti  -c -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10805 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="ESP32_DEV" -DARDUINO_VARIANT="esp32"  -DESP32 -DCORE_DEBUG_LEVEL=0 "-I/home/jonny/Arduino/hardware/espressif/esp32/cores/esp32" "-I/home/jonny/Arduino/hardware/espressif/esp32/variants/esp32" "/tmp/arduino_build_786844/sketch/esp32-fastled-webserver.ino.cpp" -o "/dev/null"
"/home/jonny/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/config" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/bluedroid" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/app_trace" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/app_update" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/bootloader_support" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/bt" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/driver" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/esp32" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/esp_adc_cal" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/ethernet" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/fatfs" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/freertos" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/heap" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/jsmn" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/log" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/mdns" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/mbedtls" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/mbedtls_port" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/nvs_flash" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/openssl" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/spi_flash" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/sdmmc" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/spiffs" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/tcpip_adapter" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/ulp" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/vfs" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/wear_levelling" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/xtensa-debug-module" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/coap" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/console" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/expat" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/json" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/lwip" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/nghttp" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/soc" "-I/home/jonny/Arduino/hardware/espressif/esp32/tools/sdk/include/wpa_supplicant" -std=gnu++11 -fno-exceptions -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlon [... cut ...]
Using library WiFi at version 1.0 in folder: /home/jonny/Arduino/hardware/espressif/esp32/libraries/WiFi 
Using library WebServer_tng-master at version 1.0 in folder: /home/jonny/Arduino/libraries/WebServer_tng-master 
Using library FS at version 1.0 in folder: /home/jonny/Arduino/hardware/espressif/esp32/libraries/FS 
Using library SPIFFS at version 1.0 in folder: /home/jonny/Arduino/hardware/espressif/esp32/libraries/SPIFFS 
Using library EEPROM at version 1.0 in folder: /home/jonny/Arduino/hardware/espressif/esp32/libraries/EEPROM 
exit status 1
Error compiling for board ESP32 Dev Module.`

per1234, 2018-04-01:
From this:
```
Using library FastLED-3.1.8 at version 3.1.6 in folder: /home/jonny/Arduino/libraries/FastLED-3.1.8
```
I can see you're using https://github.com/FastLED/FastLED. But note the documentation for this library [says](https://github.com/jasoncoon/esp32-fastled-webserver#libraries):
>Libraries
>- [samguyer/FastLED](https://github.com/samguyer/FastLED)

jasoncoon, 2018-04-01:
Also, the actual error looks to be `Invalid pin specified`.  What pin are you using?

jonny190, 2018-04-02:
Good shout @per1234  this is now letting me compile but getting:

`Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
E (20) SPIFFS: mount failed, -10025
Listing directory: /
Failed to open directory
HTTP server started
Main code running on core 1
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x400d603c  PS      : 0x00060031  A0      : 0x800d6275  A1      : 0x3ffc05a0  
A2      : 0x3ffc34d8  A3      : 0x0028800a  A4      : 0x00000002  A5      : 0x00000001  
A6      : 0x00000000  A7      : 0x00000020  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x000000ff  A13     : 0x3ff56800  
A14     : 0x3ff56800  A15     : 0x000f8023  SAR     : 0x00000007  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

Backtrace: 0x400d603c:0x3ffc05a0 0x400d6272:0x3ffc05c0 0x400d28dd:0x3ffc05e0 0x40081689:0x3ffc0610 0x4000bfed:0x00000000

`

jonny190, 2018-04-02:
good shout @per1234 ive manages to get it compiled and uploaded the Spiffs, but now looks to be in a reboot loop
`Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
Listing directory: /
  FILE: /favicon.ico  SIZE: 4286
  FILE: /index.htm  SIZE: 10733
  FILE: /css/styles.css  SIZE: 31
  FILE: /images/atom196.png  SIZE: 5469
  FILE: /js/app.js  SIZE: 13101
E (91) tcpip_adapter: tcpip_adapter is not initialized!
abort() was called at PC 0x400e1793 on core 1

Backtrace: 0x400881d8:0x3ffd3e10 0x400882d7:0x3ffd3e30 0x400e1793:0x3ffd3e50 0x400d84d9:0x3ffd3e90 0x400d49c2:0x3ffd3eb0 0x400d5074:0x3ffd3f30 0x40121ff7:0x3ffd3f60

Rebooting...
ets Jun  8 2016 00:22:57
`