jasoncoon/esp32-fastled-webserver #5: Reboot Loop

jonny190, 2018-04-14:
Hi Guys 
After getting a successful compile i'm not getting a reboot loop:
`Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
Listing directory: /
  FILE: /favicon.ico  SIZE: 4286
  FILE: /index.htm  SIZE: 10733
  FILE: /css/styles.css  SIZE: 31
  FILE: /images/atom196.png  SIZE: 5469
  FILE: /js/app.js  SIZE: 13101
E (119) tcpip_adapter: tcpip_adapter is not initialized!
abort() was called at PC 0x400e1817 on core 1

Backtrace: 0x400881d8:0x3ffd4060 0x400882d7:0x3ffd4080 0x400e1817:0x3ffd40a0 0x400d8539:0x3ffd40e0 0x400d49d6:0x3ffd4100 0x400d50b8:0x3ffd4180 0x4012207b:0x3ffd41b0

Rebooting...
`

jasoncoon, 2018-04-19:
Hmm, are you using the samguyer fork of FastLED for ESP32?  https://github.com/samguyer/FastLED

jonny190, 2018-04-23:
I wasnt up until just not but it still looks to be in a loop after changing to it
`Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
Listing directory: /
  FILE: /favicon.ico  SIZE: 4286
  FILE: /index.htm  SIZE: 10733
  FILE: /css/styles.css  SIZE: 31
  FILE: /images/atom196.png  SIZE: 5469
  FILE: /js/app.js  SIZE: 13101
E (164) tcpip_adapter: tcpip_adapter is not initialized!
abort() was called at PC 0x400e168b on core 1

Backtrace: 0x400881d8:0x3ffd3f10 0x400882d7:0x3ffd3f30 0x400e168b:0x3ffd3f50 0x400d83ad:0x3ffd3f90 0x400d49e6:0x3ffd3fb0 0x400d50c8:0x3ffd4030 0x40121eef:0x3ffd4060

Rebooting...
`

jonny190, 2018-05-23:
Can you upload a pre compiled binary ?