jasoncoon/esp32-fastled-webserver #7: Another reboot loop

edg2s, 2018-06-21:
After installing all the requirements (including forked libraries) I get this reboot loop:

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11572
entry 0x40078a5c
Listing directory: /
  FILE: /favicon.ico  SIZE: 4286
  FILE: /index.htm  SIZE: 10733
  FILE: /js/app.js  SIZE: 13101
  FILE: /images/atom196.png  SIZE: 5469
  FILE: /css/styles.css  SIZE: 31
Connecting to Il Bosco
HTTP server started
Main code running on core 1
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x400d71dc  PS      : 0x00050034  A0      : 0x40081960  A1      : 0x3ffc05e0  
A2      : 0x00000001  A3      : 0x3ff56000  A4      : 0x3ffc5510  A5      : 0x00000001  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x00000018  A9      : 0x00000001  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x8008dc1f  A13     : 0x3ffde6a0  
A14     : 0x3ffcb5e0  A15     : 0xbaad5678  SAR     : 0x00000019  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x4000bff0  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400d71dc

Backtrace: 0x400d71dc:0x3ffc05e0 0x4008195d:0x3ffc0610 0x4000bfed:0x00000000

Core 1 register dump:
PC      : 0x400e48ee  PS      : 0x00060e34  A0      : 0x8008be5c  A1      : 0x3ffd0db0  
A2      : 0x00000008  A3      : 0x00000001  A4      : 0x00000001  A5      : 0x3ffd08e8  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x3ffc5e9c  A9      : 0x3ffc5e60  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x8008d138  A13     : 0x3ffd3bb0  
A14     : 0x00000000  A15     : 0x3ffd0a60  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400e48ee:0x3ffd0db0 0x4008be59:0x3ffd0dd0

Rebooting...
ets Jun  8 2016 00:22:57
```

zombodotcom, 2018-12-14:
getting this now as well

```
10:30:24.803 -> Listing directory: /
10:30:24.871 ->   FILE: /css/styles.css  SIZE: 32
10:30:24.871 ->   FILE: /favicon.ico  SIZE: 4286
10:30:24.904 ->   FILE: /images/atom196.png  SIZE: 5469
10:30:24.904 ->   FILE: /index.htm  SIZE: 10965
10:30:24.904 ->   FILE: /js/app.js  SIZE: 13568
10:30:25.004 -> Connecting to Zombo
10:30:25.137 -> HTTP server started
10:30:25.137 -> Main code running on core 1
10:30:25.372 -> ....
10:30:25.940 -> WiFi connected
10:30:25.940 -> IP address: 192.168.0.101
10:30:26.041 -> HTTP server started
10:30:26.544 -> Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
10:30:26.577 -> Core 0 register dump:
10:30:26.577 -> PC      : 0x400d6ad5  PS      : 0x00050034  A0      : 0x40081714  A1      : 0x3ffc05e0  
10:30:26.577 -> A2      : 0x00000000  A3      : 0x3ff56000  A4      : 0x00000000  A5      : 0x3ffc5380  
10:30:26.577 -> A6      : 0x00000001  A7      : 0x00000001  A8      : 0x00000018  A9      : 0x00000001  
10:30:26.577 -> A10     : 0x00000000  A11     : 0x00000000  A12     : 0x800e0c0c  A13     : 0x3ffbba30  
10:30:26.610 -> A14     : 0x00000000  A15     : 0x3ffbbbdc  SAR     : 0x0000001f  EXCCAUSE: 0x00000005  
10:30:26.610 -> EXCVADDR: 0x00000000  LBEG    : 0x400d59ce  LEND    : 0x400d59f0  LCOUNT  : 0x00000000  
10:30:26.610 -> Core 0 was running in ISR context:
10:30:26.610 -> EPC1    : 0x400d5a8e  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400d6ad5
10:30:26.610 -> 
10:30:26.610 -> Backtrace: 0x400d6ad5:0x3ffc05e0 0x40081711:0x3ffc0610 0x400d5a8b:0x00000000
10:30:26.643 -> 
10:30:26.643 -> Core 1 register dump:
10:30:26.643 -> PC      : 0x4014ef1e  PS      : 0x00060034  A0      : 0x8008d616  A1      : 0x3ffd1590  
10:30:26.643 -> A2      : 0x00000008  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x3ffd10c8  
10:30:26.643 -> A6      : 0x00000000  A7      : 0x00000001  A8      : 0x3ffc65e8  A9      : 0x3ffc65ac  
10:30:26.643 -> A10     : 0x00000000  A11     : 0x80000001  A12     : 0x00000000  A13     : 0x00000001  
10:30:26.677 -> A14     : 0x00060420  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
10:30:26.677 -> EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
10:30:26.677 -> 
10:30:26.677 -> Backtrace: 0x4014ef1e:0x3ffd1590 0x4008d613:0x3ffd15b0
10:30:26.677 -> 
10:30:26.677 -> Rebooting...

```

jasoncoon, 2018-12-14:
Those are watchdog timer timeout resets, which means something took too long without yielding.  Try calling delay(1000/FRAMES_PER_SECOND) instead of FastLED.delay, if you're not already, reducing FRAMES_PER_SECOND, etc.  The latest code has those changes: https://github.com/jasoncoon/esp32-fastled-webserver/blob/master/esp32-fastled-webserver.ino

zombodotcom, 2018-12-14:
I am already using 
 delay(1000 / FRAMES_PER_SECOND);

same errors =(

piotrkochan, 2018-12-14:
You may try to use my Platformio fork - https://github.com/piotrkochan/esp32-fastled-webserver-platformio Just install FastLED library

jasoncoon, 2018-12-16:
@zombodotcom how many LEDs, on how many pins?  Which ESP32 board are you using?  Using latest versions of FastLED and ESP32 Arduino core?  Tried using Sam Guyer's fork of FastLED?  https://github.com/samguyer/FastLED

zombodotcom, 2018-12-16:
**How many LEDs, on how many pins?** 
25, 1 pin for testing
**Which ESP32 board are you using?**
HiLetgo ESP-WROOM-32 ESP32 ESP-32S
Node32s on the upload
Using latest versions of FastLED and ESP32 Arduino core
Completely fresh install, so updated files. 
**Tried using Sam Guyer's fork of FastLED? https://github.com/samguyer/FastLED**
Not yet.

piotrkochan, 2018-12-20:
Yesterday my ESP32 started to reboot too ;/ The weird think is that exception is thrown after very short time after short data signal pin to the leds.