jasoncoon/esp32-fastled-webserver #8: Flicker

jasoncoon, 2018-10-22:
LEDs flicker slightly.

jasoncoon, 2018-10-22:
Seems to be improved, if not completely resolved now: https://github.com/jasoncoon/esp32-fastled-webserver/commit/c6718bd4b5d7b815064de8785f65f12faca1ec76

closed 2018-10-22 by jasoncoon

pikipupiba, 2019-02-25:
I'm still getting a flicker that starts after the first 20 leds and gets more common at the far end of the strip. I have installed a logic level converter and I tried hooking the far end of the strip to the power supply as well. Neither thing has really affected the flickering at all.

jasoncoon, 2019-02-25:
Which level shifter are you using?  Are you using the latest changes mentioned above?

pikipupiba, 2019-02-25:
https://www.amazon.com/gp/product/B073D4DJDC/ref=ppx_yo_dt_b_asin_title_o00_s02?ie=UTF8&psc=1

I am pretty sure I am using the latest version. I just downloaded it 2 weeks ago.

jasoncoon, 2019-02-25:
There are very few level shifters that are known and tested to be able to handle the speed required for WS281X or SPI based (APA102, etc) RGB LEDs.  The one most used is the 74HCT245: 

More details: https://happyinmotion.com/?p=1247

pikipupiba, 2019-02-25:
Ordering new ones now. Thank you for your help!

pikipupiba, 2019-02-26:
Soooooo all that I needed to do was change the FASTLED_SHOW_CORE to 1 instead of 0 and there is no longer any flicker at all. I have no idea why this would be the case but props to u_Jem_Spencer for the fix! Maybe change this in your code?

https://www.reddit.com/r/FastLED/comments/aup7ia/getting_a_flicker_when_using_esp32_and_level/

jordanadania, 2020-05-25:
pikipupiba's solution worked for me