jasoncoon/esp32-fastled-webserver #9: Still facing issues with static files

chrisla23, 2018-11-16:
Hello,

I am still facing issues with the problem mentioned here: 

https://github.com/jasoncoon/esp32-fastled-webserver/issues/1

I have tried with both the stock SPIFFS lib and jar file and the modified ones called out in the README.

I'm using Arduino 1.87.

I am able to retrieve the /all as mentioned.

Are there any updates to the steps to make it work? 

Thanks,

-Chris

chrisla23, 2018-11-16:
> Make sure you're downloading the raw data files for libspiffs.a and
> esp32fs.jar. GitHub makes it easy to download a web page representing the
> files, instead of the actual files. I think I've fixed the links in the
> README so that's less likely now.
>
> Also, I've had some ESP32 boards where I had to upload to SPIFFS multiple
> times before it finally worked.

I had downloaded them via the zipball.

$ shasum
~/Documents/Arduino/hardware/espressif/esp32/tools/sdk/lib/libspiffs.a

f35d0918a5eda1914054d16bcb08246d0df8008a
/Users/chris/Documents/Arduino/hardware/espressif/esp32/tools/sdk/lib/libspiffs.a

$ file
~/Documents/Arduino/hardware/espressif/esp32/tools/sdk/lib/libspiffs.a

/Users/chris/Documents/Arduino/hardware/espressif/esp32/tools/sdk/lib/libspiffs.a:
current ar archive

$ shasum ~/Documents/Arduino/tools/ESP32FS/tool/esp32fs.jar

efb55039581b458df6e85c1ba30f67095c15bd29
/Users/chris/Documents/Arduino/tools/ESP32FS/tool/esp32fs.jar

$ file /Users/chris/Documents/Arduino/tools/ESP32FS/tool/esp32fs.jar

/Users/chris/Documents/Arduino/tools/ESP32FS/tool/esp32fs.jar: Zip archive
data, at least v1.0 to extract

I've tried several times at this point and tried erasing the board
with the espressif windows
tool and reflashing it. Writing the spiffs data goes very quickly. I read
in one thread that if it were working correctly it should be slower.

Thanks,

-Chris

jasoncoon, 2018-11-21:
Huh, the checksum for my libspiffs.a doesn't match that of yours:

```
shasum ~/Documents/Arduino/hardware/espressif/esp32/tools/sdk/lib/libspiffs.a
c27f3ac780f9dfb86c2f09f6155cde44ecb27d34
```

jasoncoon, 2018-11-21:
Sorry, disregard.  I had been testing with another version of that file.

So you don't get any errors while uploading to SPIFFS, but it goes very quickly?  Do you see any serial output from the sketch setup mentioning SPIFFS?  It should list the contents.

paulmneves, 2018-12-04:
Hello,

Confirming that I have this issue as well. By all appearances, it looks like the web server cannot access or serve up the files for whatever reason. I've tried using the libspiffs.a from the 'arduino-esp32' master repo as well as the 'fixed' libspiffs.a and esp32fs.jar files you recommended, neither with any success.

As I am a novice, I have not been terribly successful in debugging this further or resolving it. I would be most pleased if we could come up with a fix soon; I am trying to put some WS2811 strands on my xmas tree.

paulmneves, 2018-12-05:
@jasoncoon, thanks for the quick reply! I am indeed running these versions of these files, as verified with checksum.

I get no errors or even warnings when uploading the files to SPIFFS, and it takes all of 3-4 seconds to upload the files:

`[SPIFFS] upload : C:\Users\paulm\AppData\Local\Temp\arduino_build_443776/esp32-fastled-webserver.spiffs.bin
[SPIFFS] address: 2691072
[SPIFFS] port   : COM6
[SPIFFS] speed  : 115200
[SPIFFS] mode   : dio
[SPIFFS] freq   : 40m

esptool.py v2.5.0

Serial port COM6

Connecting........____

Chip is ESP32D0WDQ6 (revision 1)

Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse

MAC: <REDACTED>

Uploading stub...

Running stub...

Stub running...

Configuring flash size...

Auto-detected Flash size: 4MB

Compressed 1503232 bytes to 19179...

Writing at 0x00291000... (50 %)
Writing at 0x00295000... (100 %)
Wrote 1503232 bytes (19179 compressed) at 0x00291000 in 3.7 seconds (effective 3262.6 kbit/s)...

Hash of data verified.

Leaving...

Hard resetting via RTS pin...
`

jonnypage-d3, 2018-12-05:
Where do I place these files on a windows install? I can't seem to find the folders referenced anywhere. It seems the install location for the arduino ide and it's associated files is 
`C:\Users\<user>\AppData\Local\Arduino15 `
but I can't see the subfolders there.

jasoncoon, 2018-12-06:
I think on Windows, they'll go here:

C:\Users\<user>\Documents\Arduino\hardware\espressif\esp32\tools\sdk\lib\libspiffs.a
C:\Users\<user>\Documents\Arduino\tools\ESP32FS\tool\esp32fs.jar

The directories may not already exist, and may need to be created.

jonnypage-d3, 2018-12-06:
I put the files in those locations, but anything after 
`C:\Users\<user>\Documents\Arduino `
did not exist, so I had to make all the sub directories, telling me that was probably not the correct location. Didn't work regardless, still getting 
`ERR_CONTENT_LENGTH_MISMATCH`
I'll investigate further.

chrisla23, 2018-12-06:
BTW, still following this although my problems all went away when switched
from the HiLetgo Amazon special ESP32 board to the Adafruit version. It
would be preferable to have the cheaper generic board work as well, but
that seemed to be a factor for me. The cost is always a bit higher, but it
sure is worth it often to have LadyAda work through all this BS for you. :)

Thanks,

-Chris

jonnypage-d3, 2018-12-06:
@chrisla23 I have an aliexpress ESP32 board, specifically [this one](https://www.aliexpress.com/item/MH-ET-LIVE-ESP32-Development-Board-WiFi-Bluetooth-Ultra-Low-Power-Consumption-Dual-Core-ESP-32/32820820998.html) - Seems to work great within the Arduino ide as a "MH ET LIVE ESP32DevKIT" board (after installing the ESP32 in the board manager)

timbudtwo, 2019-01-19:
Also having the same issue. The site cannot be reached.
On Macos Mojave, running arduino 1.8.8
Using esp32_devkitc_v4, official silicon, from gridconnect.
Also tried using both the espressif webserver.h and ESP8266WebServer.h
Also using the samguyer fastled

Tried building with the ESP boards manually installed, and also tried it installing via boards manager. Replaced libspliffs and esp32fs, hashes "
`
efb55039581b458df6e85c1ba30f67095c15bd29 and f35d0918a5eda1914054d16bcb08246d0df8008a. `

I'm getting this in the serial output:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10088
load:0x40080400,len:6380
entry 0x400806a4
Listing directory: /
  FILE: /favicon.ico  SIZE: 4286
  FILE: /css/styles.css  SIZE: 31
  FILE: /images/atom196.png  SIZE: 5469
  FILE: /js/app.js  SIZE: 13101
  FILE: /index.htm  SIZE: 10733
Hostname: ESP32-d3a4ae30
Connecting to <redacted>
HTTP server started
Main code running on core 1
...........
WiFi connected
IP address: 10.0.1.111
HTTP server started

```

Using Angry IP Scanner 10.0.1.111 is alive but hostname does not resolve

jasoncoon, 2019-01-20:
Does it work via IP instead of hostname? http://10.0.1.111?

Can you post a screenshot of the network tab in Chrome dev tools when trying to load the page?

timbudtwo, 2019-01-22:
[image]

jasoncoon, 2019-01-22:
Does this return any data?  http://10.0.1.111/all

You should see a JSON response like this:

[{"name":"power","label":"Power","type":"Boolean","value":0},{"name":"brightness","label":"Brightness","type":"Number","value":255,"min":1,"max":255},{"name":"pattern","label":"Pattern","type":"Select","value":1,"options":["Pride","Color Waves","Rainbow Twinkles","Snow Twinkles","Cloud Twinkles","Incandescent Twinkles","Retro C9 Twinkles","Red & White Twinkles","Blue & White Twinkles","Red, Green & White Twinkles","Fairy Light Twinkles","Snow 2 Twinkles","Holly Twinkles","Ice Twinkles","Party Twinkles","Forest Twinkles","Lava Twinkles","Fire Twinkles","Cloud 2 Twinkles","Ocean Twinkles","Rainbow","Rainbow With Glitter","Solid Rainbow","Confetti","Sinelon","Beat","Juggle","Fire","Water","Solid Color"]},{"name":"speed","label":"Speed","type":"Number","value":30,"min":1,"max":255},{"name":"autoplay","label":"Autoplay","type":"Section"},{"name":"autoplay","label":"Autoplay","type":"Boolean","value":0},{"name":"autoplayDuration","label":"Autoplay Duration","type":"Number","value":10,"min":0,"max":255},{"name":"solidColor","label":"Solid Color","type":"Section"},{"name":"solidColor","label":"Color","type":"Color","value":"0,0,255"},{"name":"fire","label":"Fire & Water","type":"Section"},{"name":"cooling","label":"Cooling","type":"Number","value":49,"min":0,"max":255},{"name":"sparking","label":"Sparking","type":"Number","value":60,"min":0,"max":255},{"name":"twinkles","label":"Twinkles","type":"Section"},{"name":"twinkleSpeed","label":"Twinkle Speed","type":"Number","value":4,"min":0,"max":8},{"name":"twinkleDensity","label":"Twinkle Density","type":"Number","value":5,"min":0,"max":8}]

timbudtwo, 2019-01-22:
```
This site can’t be reached 10.0.1.111 took too long to respond.
Search Google for all
ERR_CONNECTION_TIMED_OUT
```

jasoncoon, 2019-01-22:
I think that means we're down to basic network troubleshooting then.

Can you `ping 10.0.1.111` and/or `traceroute 10.0.1.111` or `tracert 10.0.1.111`?  What's the IP address of the computer from which you're trying to access the ESP8266?

timbudtwo, 2019-01-23:
```
PING 10.0.1.111 (10.0.1.111): 56 data bytes
64 bytes from 10.0.1.111: icmp_seq=0 ttl=255 time=13.965 ms
64 bytes from 10.0.1.111: icmp_seq=1 ttl=255 time=95.142 ms
64 bytes from 10.0.1.111: icmp_seq=2 ttl=255 time=12.412 ms
64 bytes from 10.0.1.111: icmp_seq=3 ttl=255 time=366.258 ms
64 bytes from 10.0.1.111: icmp_seq=4 ttl=255 time=159.465 ms
64 bytes from 10.0.1.111: icmp_seq=5 ttl=255 time=77.372 ms
64 bytes from 10.0.1.111: icmp_seq=6 ttl=255 time=20.297 ms
```