jasoncoon/esp32-fastled-webserver #12: AP mode connection failure

Daymiin, 2018-12-15:
Set everything up, works if controlled over home wifi network.

Want to try the Access point mode because my light setup is supposed to be portable.
I can connect to the created AP and go to the webserver (192.168.0.115)
but nothing loads apart of the header and the "connecting, please wait..." message at the bottom.

Serial monitor displays the following:
"Listing directory: /
  FILE: /css/styles.css  SIZE: 31
  FILE: /favicon.ico  SIZE: 4286
  FILE: /images/atom196.png  SIZE: 5469
  FILE: /index.htm  SIZE: 10733
  FILE: /js/app.js  SIZE: 13101
Hostname: EINHORN40a4ae30
Connect to Wi-Fi access point: EINHORN40a4ae30
and open http://192.168.4.1 in your browser
HTTP server started
Main code running on core 1
..........................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................."
with the points going endlessly.
No idea whats causing this.
Help would be nice.

jasoncoon, 2018-12-16:
I haven't done much testing with AP mode lately.  Where'd you get the IP address 192.168.0.115 from?  Tried 192.168.4.1, as noted in the serial output?

Daymiin, 2018-12-16:
192.168.0.115 is the IP that works in the normal network mode, 192.168.4.1 isn't addressable in AP or normal mode.
And the header is loading at the 192.168.0.115 address so I guessed it would be the right one to go to.
The 192.168.4.1 address ist just a static serial.print which I think can change depending on the hardware address of your microcontroller.

timbudtwo, 2019-01-24:
I am having the issue of it not giving my device a dhcp address I think? I tried setting this up in AP mode since standard wifi was not working T^T. I get one of two errors:
`E (27915) event: mismatch or invalid event, id=63
E (27915) event: default event handler failed!
.dhcps: send_offer>>udp_sendto result 0`

Or 
```ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10088
load:0x40080400,len:6380
entry 0x400806a4
Listing directory: /
  FILE: /favicon.ico  SIZE: 4286
  FILE: /css/styles.css  SIZE: 31
  FILE: /images/atom196.png  SIZE: 5469
  FILE: /js/app.js  SIZE: 13101
  FILE: /index.htm  SIZE: 10733
Hostname: BlackHat-d3a4ae30
Connect to Wi-Fi access point: BlackHat-d3a4ae30
and open http://192.168.4.1 in your browser
HTTP server started
Main code running on core 1
.........................................................................................................................................................................................................................E (35480) event: mismatch or invalid event, id=63
E (35481) event: default event handler failed!
...............................................................................................................................................................................................................................................................................................................................................................................```

pikipupiba, 2019-02-25:
It's getting stuck in the handleWeb() function inside of web.h. It's waiting for the status to be WL_CONNECTED but I'm not sure that will ever be the case in this situation. What do we need to change here to let it escape the loop?

jordanadania, 2019-06-26:
Seems like an easy thing to test