jasoncoon/esp32-fastled-webserver #15: xTaskNotify - Assert Failed!

pikipupiba, 2019-02-22:
I get the sketch uploaded to my esp32 but I get the following error that resets the board constantly.
https://imgur.com/a/TznnZIZ

pikipupiba, 2019-02-22:
omg I updated FastLED and it works now... I am terribly sorry.

closed 2019-02-22 by pikipupiba

jasoncoon, 2019-02-22:
No problem, glad it works now!  Was just investigating and typing up some questions for you.  :)

pikipupiba, 2019-02-22:
I was actually looking at the other issues and then when I went to check if I had the right branch of FastLED I saw the update button. Turns out no one else was having the problem because it was literally the first thing you are supposed to check... I honestly feel very bad about the whole situation and how long it took me to find that. It did make me very happy to hear from the man himself though! I'll try to work a little smarter instead of harder from here on out!