jasoncoon/esp32-fastled-webserver #18: Wrong Color Order

jordanadania, 2019-06-25:
FastLED.addLeds<WS2812B, 23, GRB>(leds, NUM_LEDS);

FastLED.addLeds<WS2811, 32, RGB>(leds, NUM_LEDS);

With that code, the WS2812B strip runs correctly, but the WS2811 code swaps the Red and Green. Switching the WS2811 to GRB has no effect.

Putting the WS2811 line before the WS2812B gives opposite results. The WS2811 strip will run correctly, but the WS2812B strip would have Red and Green swapped. Switching the WS2812B to RGB has no effect.

jordanadania, 2019-06-25:
I am aware this isn't an issue with Jason Coon's code.

jasoncoon, 2019-06-25:
Sorry, I'm not sure.

Since this is a general FastLED question, I'd recommend posting it on the official forums (reddit): https://www.reddit.com/r/FastLED/

jordanadania, 2019-06-25:
Done. Thank you for the advice.

jasoncoon, 2019-06-25:
For reference: https://www.reddit.com/r/FastLED/comments/c5gqf7/esp32_fastled_wrong_color_order/

closed 2019-06-25 by jasoncoon