jasoncoon/esp32-fastled-webserver #20: Lots of warnings when building

henrygab, 2019-08-22:
There are a number of warnings building this repository.  For example:

### Missing [field initializers](https://en.cppreference.com/w/cpp/language/default_initialization)

In fields.h, while FieldList is an array, the array elements are each a POD struct of type Field, which has no non-default constructor.  Therefore, according to the specification:
> The effects of default initialization are: 
> …
> otherwise, nothing is done: the objects with automatic storage duration (and their subobjects) are initialized to indeterminate values. 

### Comparisons always false due to limited range of data type

Simply put, the `long` result from `string.toInt()` was placed into an `uint8_t`, prior to clamping the value.  This change stores the conversion result in a `long` to perform the clamping prior to storing the value in the `uint8_t` field.

### "warning: extra tokens at end of #include directive"

No semi-colons after #include "file.h"

### "error: multi-line comment [-Werror=comment]"

Don't allow the last character of a comment line to end in a backslash character.

### "warning: 'typedef' was ignored in this declaration"

The `typedef` of `Field` is in the wrong spot.

marcmerlin, 2020-03-01:
@jasoncoon  are you able to include this patch? I'm seeing the same errors and for me such warnings count as errors.

jasoncoon, 2020-03-05:
Sorry for the massive delay, thank you for the PR!

PR #21 (2020-03-05): Update to build cleanly
Fixes #20
Fixes #10

### Use `webserver.h` instead of `ESP8266WebServer.h`

Makes this build without dependencies.

### Fix `Missing field initializers` warnings.

In fields.h, while FieldList is an array, the array elements are each a POD struct of type Field, which has no non-default constructor. Therefore, according to the specification:
> The effects of default initialization are:
> …
> otherwise, nothing is done: the objects with automatic storage duration (and their subobjects) are initialized to indeterminate values.

### Fix `Comparisons always false due to limi [... cut]
closed 2020-03-05 by jasoncoon